lysmarine/lysmarine_gen #1: Openplotter integration tread

FredericGuilbault, 2019-04-12:
Here all the notes and links for the ongoing process of integrating openplotter into lysmarine.

FredericGuilbault, 2019-04-12:
The issues I have reported to sailoog are going to be fixed in the 2.x.x version.

FredericGuilbault, 2019-04-12:
There are pull request open

mgrouch, 2020-11-25:
Might be openplotter is not needed.
It was very useful for sometime. 
It’s just bunch of scripts for setting things up. If we recognize know hardware it might be not needed. 
At least I would hold on on integrating

FredericGuilbault, 2020-11-25:
yeah, it's mainly there cuz it's a possibility to implement and I have no plan to do it my self.   They still don't do things "the linux way" although they made great improvement since their first version and that will break stuff.

closed 2020-12-14 by FredericGuilbault