lysmarine/lysmarine_gen #16: Known problems regarding fbpanel

FredericGuilbault, 2019-04-12:
Dirty fix for task icon size: https://github.com/aanatoly/fbpanel/issues/46

Configuration window does not have a scrolling bar. https://github.com/aanatoly/fbpanel/issues/47

mgrouch, 2020-10-13:
Try not using bitmap icons at all. Pick a theme with all SVG icons.
And might be modify the code of fbpanel to display SVG.
And for enforcement make fbpanel accept only svg icons.
Just an idea

FredericGuilbault, 2020-10-27:
IDK Cpp much, I hacked the minimum to make fbpanel work but I wouldn't be able to make such changes. A lot a people are pushing PR on the original repo. But the maintainer seem gone and one seem to fork it to become the new maintainer.

mgrouch, 2020-10-27:
This fork seems to be active

https://github.com/akozlins/fbpanel

Might be better use that fork as a base?

mgrouch, 2020-11-13:
I think fbpanel set up just need to be simplified.
All it needs to show is just three buttons like in android. Triangle/Circle/Square. In the middle.
I think fbpanel should be on right as most people are right handed.

It's ok to have status icons on top and button. Like network icon, on screen keyboard icon, etc

This way we just workaround all other fbpanel issues.

FredericGuilbault, 2020-11-15:
Before the nativfier desktop background. Lysmarine  looked way more like a desktop OS So desktop alike panel bar was kinda needed.  But I always wanted it to be like an embedded dedicated onboard computer. Not just a Linux desktop with a list of pre-installed software packages, (this already exist in the form of navigatrix and openplotter and don't need a 3rd player. ...at least since OP2).

While trying to unify the web/remote experience with the desktop one I came to go full web on the desktop because the other way was not much possible (VNC/rdp sucks)   The only other option i know is Apache guacamole but was still hacky as VNC and heavy on resources.  

Now the need of fbpannel is slowly going away, But there still a list of softwares that can't be put as web resource, Mainly Opencpn, xygrib, and the terminal&filemanager for maintenance reasons. The avnav project is quite close to been able to avoid desktop tool at all. But they achieve it to the expense of cutting down a too much customization & management capability.

I don't personally like fbpanel. And it's mostly a dead project. But it's doing the job so far.  I think the 3 event could be a good thing, If I recall openbox can listen to events send by xdotool in the regard of window management.   But this would mean writing a plugin for fbpanel ???

mgrouch, 2020-11-15:
I think you can just launch any app like OpenCPN etc from lysmarine desktop web app. Just from the web server side. Still need a way to switch between apps on desktop. I think fbpanel needs to go. It’s buggy with the touch and unmaintained. Might be use tint2.

FredericGuilbault, 2020-11-15:
switching to tint2 seem more realistic considering the available resources. I remember testing it, but I don't remember why I choose fbpanel instead.  

Writing webbased desktop management bind in nodejs is a complete project in itself. Unless a nodejs module already exist....

mgrouch, 2020-11-16:
https://github.com/mixu/nwm

mgrouch, 2020-11-16:
Or just use lxpanel instead. We will be consistent with default for raspberry pi and openplotter.

FredericGuilbault, 2020-11-18:
mixu/nwm   Last commit on Sep 9, 2015    <--- this is like 9000 years old in the nodejs world :p , I doubt this package even install.

My memory is slowly coming back and one of the reasons why I decided to use FBpanel was that I can use right click to manage windows and therefor I could save precious screen size by disabling the window decorations.

mgrouch, 2020-11-20:
I simply replaced fbpanel with
budgie. Looks awesome. 
Lots of shades of grey. Still less than 50 I think

FredericGuilbault, 2020-11-24:
From memory, budgie was not a lightweigth DE.

I made a quick google and came across this vid and it looks like it's not. ( for what a youtube vid worth as scientific proof ) 

Is the taskbar inbedded in the DE ? Or it's a stand alone component.

mgrouch, 2020-11-24:
You can simply try.

sudo apt install budgie-desktop

and start it

budgie-desktop
(for DE)

or

budgie-panel
(for just panel)

It's lightweight and awesome.

Thanks

FredericGuilbault, 2020-11-24:
> It's lightweight and awesome.

https://www.phoronix.com/scan.php?page=article&item=ubu-1704-desktops&num=2
https://www.phoronix.com/scan.php?page=article&item=ubu-1704-desktops&num=3

It's a 2007 review but it's roughly 2x openbox....

mgrouch, 2020-11-24:
It’s 2020 now

mgrouch, 2020-11-24:
Try newly built image 

https://cloudsmith.io/~bbn-projects/repos/bbn-repo/packages/?q=lysmarine

It’s looking interesting

mgrouch, 2020-11-25:
No more fbpanel. 
Close it ;)

mgrouch, 2020-11-25:
A far as memory is concerned. It might take twice more memory, but it serves 12 more in functionality.  
Plus it’s written in Vala which is a big plus. Vala is c-sharp like language with reference counting translatable into C before compiling. Strong typed, much safer memory model, portable as C, performant as C. I’d take that even for longterm

FredericGuilbault, 2020-11-25:
Im not hot for budgie, 

Before finally opting for openbox dwm and i3 where my first choices. At the time, rpi2 was still massively used and 4.3 inch screen where common (no real need of window manager when all you can do is full-screen one app at the time.). 
---

Now yes it's 2020 and we have fancy RPI4,  and the PI foundation is kida trying to do things maybe better... But there still plenty of slow SBC around that are probably more suited to be board computers then a RPI.

And as remote connection is a must right now, (and VNC sucks)  The webapps are the way to go.   Since lysmarine_8.x im trying to leave desktop apps .  In a perfect world lysmarine would boot like avnav.   A simple Xorg server that do only one thing: launching a web browser full screen. 

For those reasons I'm not excited by budgie, It push toward making low-end processors useless e-waste, and not helping in the direction of having a unified UX locally and remotely. 
---

This said, I don't want to be a friction that kill progress and contributions and all of this is only my personal brokeass-oldschool-linux-poweruser point of view. 

So finally, I suggest that you create a new stage for budgie, let's say `7-desktop-budgie` VS `6-desktop-openbox` So we will be able to build one OS  for low-tier SBC and one for modern SBC. That will also be the occasion to relocate configs that are inside openbox but should be set regardless of the DE.

FredericGuilbault, 2020-12-03:
For my part, the branch that I will use to make the change from fbpanel to lxpanel is here: https://github.com/lysmarine/lysmarine_gen/tree/improvement/lxpanel

mgrouch, 2020-12-03:
I think lxpanel-gtk3 would be better option. Gtk2 has issues with touchscreen. Someone made gtk3 version.

mgrouch, 2020-12-03:
https://sourceforge.net/p/lxde/bugs/773/

mgrouch, 2020-12-03:
Or LXQt version 

https://en.wikipedia.org/wiki/LXQt

mgrouch, 2020-12-03:
Tried lxqt version and it looked slow

FredericGuilbault, 2020-12-03:
HO! I thought it where gtk3 but haven't checked. Got me before I waste a lot of time on this.  Thanks. 

After reading.

The devs seem to be pissed at gtk3 ....like many other devs. And prefer to move to Qt instead of gtk3 
https://wiki.lxde.org/en/LXDE:Questions#Plans_to_migrate_to_GTK3.2B.3F
___

At the same time, from what I found on arch BBS,   lxpanel do compile with gtk3 without patch,  It's just that debian don't do it yet (probably to stay consistent with the rest of the LXDE ).  
https://github.com/archlinux/svntogit-community/blob/master/lxpanel-gtk3/repos/community-x86_64/PKGBUILD

And looking at lxpanel repository there been commit made to support gtk3. 
__
So, lxpanel support gtk3, but debian don't build it that way. Never less the devs are about to drop support for the whole gtk2/3 line anyway. So on the long run lxpanel is not a good choice.

___
On the other hand.   lxqt-panel have many dependencies, and strangely it includes kde components... and xml parsers. 
```
Depends: lxmenu-data, lxqt-policykit, libasound2 (>= 1.0.16), libc6 (>= 2.29), libdbusmenu-qt5-2 (>= 0.3.2), libkf5solid5 (>= 4.97.0), libkf5windowsystem5 (>= 5.6.0+git20150112.0023+15.04), liblxqt-globalkeys-ui0 (>= 0.8.0), liblxqt-globalkeys0 (>= 0.9.0), liblxqt0 (>= 0.11.0), libpulse0 (>= 0.99.1), libqt5core5a (>= 5.12.2), libqt5dbus5 (>= 5.0.2), libqt5gui5 (>= 5.8.0) | libqt5gui5-gles (>= 5.8.0), libqt5widgets5 (>= 5.11.0~rc1), libqt5x11extras5 (>= 5.6.0), libqt5xdg3 (>= 3.1.0), libqt5xml5 (>= 5.0.2), libsensors5 (>= 1:3.5.0), libstatgrab10, libstdc++6 (>= 5.2), libsysstat-qt5-0 (>= 0.2.0), libx11-6, libxcb-xkb1, libxcb1, libxcomposite1 (>= 1:0.3-1), libxdamage1 (>= 1:1.1), libxkbcommon-x11-0 (>= 0.5.0), libxkbcommon0 (>= 0.5.0), libxrender1
```
So not much a clean lightweight independent solution. 
___

tint2 is also gtk2
___

But vala-panel seem to be clean,small and gtk3 

I remember you mentioning it, was it doing the job ?

mgrouch, 2020-12-04:
Budgie desktop is what I picked after all. Interestingly lxterminal is gtk3
pcmanfm is gtk2. So I’m about to replace it with 
Thunar

mgrouch, 2020-12-04:
Qt stuff is slow. also I need some kind of task manager. Lxtasks is gtk2

FredericGuilbault, 2020-12-10:
This is how arch compile lxpanel to gtk3
https://github.com/archlinux/svntogit-community/blob/packages/lxpanel-gtk3/trunk/PKGBUILD

mgrouch, 2020-12-11:
I think I’m well ahead in my fork. It’s really starting looking great. Well balanced as desktop and tablet OS. 
I did try Android too. I think Lysmarine OS in my fork is coming quite competitive for a cockpit chartplotter. I’ll attach a picture

mgrouch, 2020-12-11:
I’ve posted a video to give an idea how it works on Facebook lysmarine forum

mgrouch, 2020-12-11:
A lot of stuff works nice. 
Most of it. 
Today I went via printing, samba (seeing shared windows directories on network). Touch UI is responsive. 
Menus are mostly done. GPS devices are recognized on plugging in. One AIS too. dAISy AIS. I just don’t have anything else to test. Music YouTube and TuneIn work out of the box. Both in YouTube and MusicBox apps. Facebook works. Games are chess and card game preference. WiFi works. OnScreen keyboard works. 
Email should work. I haven’t tried yet. 
Still missing some icons. Menus are set up fine. Bluetooth needed extra testing. file browser should be ok. task manager is there but it’s gtk2.

FredericGuilbault, 2020-12-12:
Since 52ea98a1e651a41069d52d1b90e04f104d47bae5 Lysmarine package lxpanel-gtk3 and pcmanfm-gtk3 
There is still tiny eyes candy to tweak but that's overall a sufficient improvement. 

This close the issue for the original topic that was about fbpanel problems. 

___

Now the other question that remain open in this tread is budgie integration. As said, I won't do it personally, now the question is will you do it ? or we close this issue ? 
___

On a more general note. I'm tankful for all the testing reviews and idea you bring on lysmarine project and honoured that you choose lysmarine as your base for your project. You bring some quite good inputs and motivate me to work more on this project :heart: . But for the rest, honestly im confused lol. On one side you act like if you were participating in the project. But at the same time you have hard fork the project and are not contributing any code.  This, while you post on lysmarine FB group and here about things that i must say, don't relate to lysmarine at all.       Now, Im curious to hear you aboot where you are going with your repo and how you see it in relation to lysmarine project & codebase ?

mgrouch, 2020-12-22:
Why is this still open?
fbpanel is gone now

FredericGuilbault, 2020-12-22:
Sorry. it's still open due to this issue : 

https://github.com/lxde/lxpanel/issues/23

mgrouch, 2020-12-22:
> Since 52ea98a Lysmarine package lxpanel-gtk3 and pcmanfm-gtk3
> On a more general note. I'm tankful for all the testing reviews and idea you bring on lysmarine project and honoured that you choose lysmarine as your base for your project. You bring some quite good inputs and motivate me to work more on this project ❤️ . But for the rest, honestly im confused lol. On one side you act like if you were participating in the project. But at the same time you have hard fork the project and are not contributing any code. This, while you post on lysmarine FB group and here about things that i must say, don't relate to lysmarine at all. Now, Im curious to hear you aboot where you are going with your repo and how you see it in relation to lysmarine project & codebase ?

I want my code to be merged into lysmarine at some point. I have few things to finish and I do keep in mind how it can be merged. But it also depends on you if you really want to take it. Realistically your build scripts are more advanced than mine so they would stay. And everything under install-scripts could be completely from my fork. After that we would need to fix few things of aftermath of merge. I could test it on arm64 rpi4. 
Might be even on arm32. What do you think of that plan?

mgrouch, 2020-12-30:
I’ve brought my fork into decent shape. What do you think about replacing install-scripts directory in
your code complete with install-scripts directory from my fork?
I was following your commits in your install-scripts so mine is basically superset. other directories in your code do not need to change really
(except increasing image size) and might be some small issues which will be uncovered by automated build on CircleCi. 

Thanks

FredericGuilbault, 2020-12-30:
> But it also depends on you if you really want to take it.  

Not everything. But most of it yes, Sadly, there are few things that don't fit in my vision of lysmarine.

I realize that I have never written down what are the vision/features/objectives/market/philosophy of this project and what make it reverent although there is already 3 others projects that try to achieve the same objective. (openploter, avnav-os and navigatrix). I should sit down and do this in priority, it should have been done at the beginning.  

But for sure, ATM for every feature request that are open here, I will welcome a Pull request (one per feature). I understand that at this point it will be a lot of work to split your monolithic hard-fork into features that does not conflict between repository. But I still think it's the best way to do it. Doing so will stop the duplication of the maintenance task between you and me and prevent polution of the main repository with all the dev workflow. The same way that I always try to contribute to other people projects, so I do not have to maintain the changes and updates after,  Plus it can profit the whole community of users. Not just the people that use lysmarine. 

For the rest, The things that are irreconcilable in my view. idk what to do with this.... I surely feel bad to turn you down, On the other hand this is part of the job of the benevolent dictator to triage what serve the project.    For sure, There can't be 2 projects with the same name and if you start a rebranding of lysmarine. I would see it as a failure to collaborate. The same way lysmarine itself is a personal failure of getting my PR and linux sysadmin views/good practices integrated in openplotter.

mgrouch, 2020-12-30:
It’s just easier to replace one directory and later in your branch to remove what you do not like to have there. I do not think I will have time for each feature PR nor I see effort/value added ratio being good here.  There was just way too many tiny separate issues in very small code base. What I could do is probably rebase my fork to yours do one delete commit and do one commit with my stuff (install-scripts only) and deliver it as one PR. This way we will be few commits apart in our code bases. 

For now I’m presenting it as Lysmarine with BBN Edition suffix to avoid confusion. It also had more issues fixed than it was reported. 
Have you ever had a chance to run it on raspberry pi 4 with some touchscreen? I’d recommend trying latest beta image I’ve made today. 
Which things do not fit your vision?
It would help. 
It’s good that there are 3 other projects. If they all opensource they could incorporate best parts and cross pollinate. 
I didn’t have much of a vision for this project I just went with a flow what I could do in shortest time to make it more or less complete product,
with focus on cockpit touchscreen but not only, and based on experience of using openplotter. 
I think mission statement might be counterproductive cause it can change every day, depending on experiments and constant re-evaluation of efforts for different approaches. 
Thanks

FredericGuilbault, 2020-12-31:
> It’s just easier to replace one directory and later in your branch to remove what you do not like to have there

Easier for you, As  you are pushing on me the task of making sense, sort and understanding the reason behind every change of a 1K lines patch. thx, but no thx.    ...I have a better way to do this then a merge if I have to do it manually my self.

Hard forking + developing/testing on the master branch was a poor decision. I said it a few times. Now the technical debt it raised is so high that neither you or me want to it payback. ¯\_(ツ)_/¯

> Have you ever had a chance to run it on raspberry pi 4 with some touchscreen
I don't have the Rpi4 yet. I tried with rpi3 it booted to a blank screen. I did not investigate.

FredericGuilbault, 2021-02-13:
Lysmarine now use awesomewm which have its own task bar capability.

closed 2021-02-13 by FredericGuilbault