lysmarine/lysmarine_gen #2: Sound dosen't work

FredericGuilbault, 2019-04-12:
The sound work, The problem is, HDMI capture it.

mgrouch, 2020-10-13:
Most likely not a bug. It depends whether user wants output on HDMI or AUX.

amixer cset numid=3 1

switches to AUX

FredericGuilbault, 2020-10-27:
`amixer cset numid=3 1` did not fix it. 
 raspi-config did not helped neither. 

I hear the sound when I do:

`/opt/vc/src/hello_pi/./hello_audio.bin 0`

But not with: 

`aplay /usr/share/sounds/alsa/Front_Center.wav`

mgrouch, 2020-10-27:
Probably volume somewhere is set to 0

mgrouch, 2020-10-27:
Or not initialized volume

mgrouch, 2020-10-30:
Or user needs to in audio Unix group

mgrouch, 2020-11-12:
I think I’ve made it work. 
Check my latest commits into my fork of lysmarine-gen

FredericGuilbault, 2020-11-15:
That's so nice :D  This is a great news. 

If you want to make a Pull request and concerning this specific issue.  ill merge it in happiness. 

 PulseAudio Volume Control  probably safe his setting with gconf or dconf. These are database storage and it's kinda hard to add stuff in them before the first boot. 

To there is the hack I made to add onboard config in the database on the first boot :
I use CLI command to add the .dconf file to the database on the first boot of openbox, then i use a sed command to comment the line right after it been executed. (and also comment the line of the sed command it self.)

https://github.com/lysmarine/lysmarine_gen/blob/master/install-scripts/6-desktop/2-onboard.sh

There is a nice GUI tool to explore the structure of dconf config. I forgot his name. It helped me a lot to write the .dconf files.

mgrouch, 2020-11-17:
Also this

Some people have asked how they can switch the audio output from the command line without using the desktop. To do this, you will need to create a file called .asoundrc in your home directory; ALSA looks for this file to determine which audio device it should use by default. If the file does not exist, ALSA uses “card 0” – which is HDMI – as the output device. If you want to set the headphone jack as the default output, create the .asoundrc file with the following contents:

defaults.pcm.card 1
defaults.ctl.card 1
This tells ALSA that “card 1” – the headphone jack – is the default device. To switch back to the HDMI output, either change the ‘1’s in the file to ‘0’s, or just delete the file.

FredericGuilbault, 2020-11-24:
I did not read the 424 commits but I found these 3 

https://github.com/bareboat-necessities/lysmarine_gen/commit/446b2f8a75cced5ebd23c9e3e7939dc5c1e50478

https://github.com/bareboat-necessities/lysmarine_gen/commit/9a628a0c7507df87a1dd3144f7a29e239a93f755

https://github.com/bareboat-necessities/lysmarine_gen/commit/b8527fc2fd75d1b713a70759ccd748c5129614e5

mgrouch, 2020-11-25:
Done in my fork

FredericGuilbault, 2020-11-29:
This one is also related: 
https://github.com/bareboat-necessities/lysmarine_gen/commit/325437f9484700553506a2373390e3066a06302d

FredericGuilbault, 2020-11-30:
This merge fix it :  04267b7a0934ffcf72b2fff3d373de69d1e36d4f

Thx for the help :D

closed 2020-11-30 by FredericGuilbault