lysmarine/lysmarine_gen #22: pcmanfm missing icons

FredericGuilbault, 2019-04-17:
This is a duplicate of : 

https://github.com/MichaIng/DietPi/issues/1558

solution 
```
wget http://ftp.us.debian.org/debian/pool/main/p/pcmanfm/pcmanfm_1.2.5-3_armhf.deb
dpkg -i *.deb
```

FredericGuilbault, 2019-04-26:
- Using debian package don't fix
 - Using dietpi package don't fix
-  `dbus-launch` is not running, killing all dbus process make openbox crash

FredericGuilbault, 2019-04-27:
Solved with a work around

closed 2019-04-27 by FredericGuilbault

MichaIng, 2019-08-01:
@FredericGuilbault 
I was reviewing this issue. The Debian (=DietPi) package does not solve the missing icons in your case? What was the workaround you applied?

Also would be good to know if this is still an issue on Raspbian Buster.

FredericGuilbault, 2019-08-01:
@MichaIng  I guess I got lazy linking the actual commit: 

https://github.com/lysmarine/lysmarine_gen/commit/198946e38225cfe4bdae24052d96ac079e6be3ab

From what I remember, I just have hidden the problematic parts via config files. ...dirty workaround.

I haven't managed to boot lysmarine with dietpi Buster yet. so IDK if the bug is still there.

FredericGuilbault, 2019-08-01:
Maybe the package  `gnome-themes-standard` have to do something with it (vague memories)

MichaIng, 2019-08-01:
@FredericGuilbault 
Would be actually a nice solution if just some icon page would be missing that is pre-configured with Raspbians pcmanfm version, e.g. https://packages.debian.org/stretch/adwaita-icon-theme or something.
I will test the next days, have a fresh DietPi Buster SDcard from image creation around to hot-plug&play 😉.