lysmarine/lysmarine_gen #26: add music server (mopidy)

mgrouch, 2020-10-13:
See https://bareboat-necessities.github.io/my-bareboat/#_music_player

It needs its own unix user account
and registers are a service

mgrouch, 2020-11-10:
Added in my fork. 
No yet support for Spotify. 
Needs more config for YouTube music.

mgrouch, 2020-11-16:
Still needs more work to support pulseaudio sound mixer.

FredericGuilbault, 2020-11-18:
I would take a pull request concerning this specific feature. That would be a nice feature.

mgrouch, 2020-11-18:
Some nice tips here

https://www.raspberrypi.org/forums/viewtopic.php?t=274113

if you installed chromium, make sure its "rpi-chromium-mods" thats the optimized version.
add "chromium-browser --headless" to startup, preloads chromium, shortens first startup lag.

mgrouch, 2020-11-18:
Is there any reason you are using pulseaudio as opposed to just alsa?
Pulseaudio adds additional configuration hurdles.

FredericGuilbault, 2020-11-18:
No real reasons.

FredericGuilbault, 2020-11-24:
I think mopidy should be somewhere in the `4-server` stage as it's web based and can easily be used without a screen attached to the RPI

mgrouch, 2020-11-25:
True

mgrouch, 2020-11-25:
Same with avnav and ppp-modem for iridium

mgrouch, 2020-11-25:
Done in my fork

FredericGuilbault, 2021-01-12:
Done in my fork

closed 2021-01-12 by FredericGuilbault