lysmarine/lysmarine_gen #27: add right click emulation via touch for touchscreens

mgrouch, 2020-10-13:
See

https://bareboat-necessities.github.io/my-bareboat/#_right_click_emulation_with_long_touch

Thanks

mgrouch, 2020-10-24:
Binaries for evdev-rce

Debian:
https://cloudsmith.io/~bbn-projects/repos/bbn-repo/groups/

Ubuntu
https://launchpad.net/~bbn-projects/+archive/ubuntu/bbn-ppa

All processor architectures and current distros

mgrouch, 2020-11-03:
I've made a new (better) package and moved it into dedicated repo

https://cloudsmith.io/~bbn-projects/repos/bbn-rce/packages/

mgrouch, 2020-11-03:
I've integrated evdev-rce debian package install into circleci lysmarine image build and tested it.
it works very nice. see my fork for changes.

FredericGuilbault, 2020-11-18:
Noticed 
:heart:  [image] :heart:

FredericGuilbault, 2020-11-19:
@mgrouch I im aboot to add your launchpad PPA of evdev-rce into lysmarine. 

But before, I was trying to understand what I'm adding exactly.  If I get it right, it's a fork of a fork of a project published by PeterCxy. By default, contributions should be favoured to forks but there is also many good reasons to fork a project. On the other hand, I found it hard to see what improvement you made on this repo. I get that your workflow involve making a commit for each test you do. May I suggest you to try GitKraken. I think I tried every graphical git applications out there and it's the only one that truly made me stop using the CLI and be proficient with branches and multi-origin management. Its one of the rare non-FOSS software I use.

Also, have you tried to make a PR to PeterCxy?  Just to be sure that he do not maintain his repo anymore. So that would justify the forking for me.

FredericGuilbault, 2020-11-19:
[image]

mgrouch, 2020-11-19:
Use my fork for now. 
I’ll try to contact original author

FredericGuilbault, 2020-11-23:
The work is happening in this branch: 
https://github.com/lysmarine/lysmarine_gen/tree/Feature/evdev-rce

FredericGuilbault, 2020-11-23:
@mgrouch your launchpad PPA need to be signed with your PGP key. (and have your key publicly available on a key registry) apt-get is complaining about it.

mgrouch, 2020-11-23:
Which one of my PPAs?

FredericGuilbault, 2020-11-23:
https://launchpad.net/~bbn-projects/+archive/ubuntu/bbn-ppa

mgrouch, 2020-11-23:
You can use cloudsmith one right now. 

wget -q -O- https://dl.cloudsmith.io/public/bbn-projects/bbn-rce/cfg/gpg/gpg.540A03461CECBA19.key | apt-key add -

deb https://dl.cloudsmith.io/public/bbn-projects/bbn-rce/deb/debian buster main
deb-src https://dl.cloudsmith.io/public/bbn-projects/bbn-rce/deb/debian buster main

While I check launchpad

mgrouch, 2020-11-23:
My key is there. You just need to get it from ubuntu key server

 apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 24A4598E769C8C51

mgrouch, 2020-11-24:
Kplex build is available too now on launchpad bbn-kplex

FredericGuilbault, 2020-11-24:
Thx, it works now with the key.

mgrouch, 2020-11-25:
Totally Done in my fork

FredericGuilbault, 2020-12-02:
If you want to to be the maintainer of  https://github.com/bareboat-necessities/evdev-right-click-emulation
 you should enable the issue tab.

FredericGuilbault, 2020-12-11:
https://github.com/bareboat-necessities/evdev-right-click-emulation/issues/2

https://github.com/bareboat-necessities/evdev-right-click-emulation/issues/1

FredericGuilbault, 2020-12-22:
No, this is still broken in lysmarine, and as the maintainer/packager you still have a valid issue to solve.  You wanted to fork that guy code, and you wanted me to integrate it. I must go throu the whole process of making a good quality integration OR roll back and remove evdev-rce.     But this can't be dropped in the middle of the road.

FredericGuilbault, 2021-01-12:
Done in my fork

closed 2021-01-12 by FredericGuilbault