lysmarine/lysmarine_gen #28: Please use UbuntuFonts on desktop (menus, forms, etc) They are much more readable on small touchscreens in a cockpit

mgrouch, 2020-10-13:
Please use Ubuntu Fonts on desktop theme (menus, forms, etc) They are much more readable on small touchscreens in a cockpit
(than the fonts raspberry OS is using)

sudo apt-get -y install fonts-ubuntu

And the default theme should be dark like on most smartphones.

Thanks,
--MG

FredericGuilbault, 2020-10-27:
fonts-ubuntu is a non-free package under debian, I have no problem with this, it only means that implementing this will include modifying the apt source.list 

---

breeze-gtk theme would be better than what I use ATM. But there is something that don't show right in Opencpn. I think This must be fixed before the white square is white on white. 
[image]

---

mgrouch, 2020-11-25:
Sans is a good font
Just use it consistently

closed 2020-11-25 by mgrouch