lysmarine/lysmarine_gen #29: QtVlm intead of packaging it, package a script which makes it simple for a user to install it

mgrouch, 2020-10-13:
This way no licensing issues. User will have to accept the license on a first run of QtVlm.

Thanks

FredericGuilbault, 2020-10-27:
Does QtVlm provide something that no other software can do ? 

Im not very comfortable with the idea of helping adoption of non-foss projects if there is FOSS solutions... specially if they don't pay me lol.

mgrouch, 2020-11-25:
I guess just dropping a script into 
/home/user
directory would do it
https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/qtvlm/rpi_qtvlm_install.sh

script and config

https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/qtvlm-conf/qtVlm.ini

See Here: https://bareboat-necessities.github.io/my-bareboat/#_qtvlm

FredericGuilbault, 2020-11-26:
> /home/user 
/usr/local/bin/ would respect linux FHS

 > /home/pi/.qtVlm/qtVlm.ini
There is no user named `pi` in lysmarine. hardcoded user will make the script fail.

FredericGuilbault, 2020-12-11:
I have read this tread agan https://forum.openmarine.net/showthread.php?tid=794  and this confirmed my choice. 

I will not put time on this. This said, I would not reject a Pull Request for an install script.  

@mgrouch to you want to do it or we close this issue ?

mgrouch, 2020-12-22:
I’ve done it in my fork. Worked only for arm32. They do not have arm64.

closed 2020-12-22 by mgrouch