lysmarine/lysmarine_gen #30: UDP vs TCP for NMEA

mgrouch, 2020-10-13:
UDP I think would be a better choice for LAN NMEA network traffic than TCP.
UDP is connectionless so it restores automatically, while I've seen 
bugs in SignalK or something else where it wouldn't restore lost TCP connections.

Thanks,
--MG

FredericGuilbault, 2020-10-27:
Hummmm, that's a good question. 

On the other hand, A big part of the UDP design is: if anything go wrong, Drop the package and move forward.   

This mean causing dataloss just to keep in sync( in terms of milliseconds) with the emitter. 

But really, I'm not knowledgeable enough in this domain to have an opinion on what would be the best.

FredericGuilbault, 2020-10-27:
I asked the question signalk usergroup 
https://groups.google.com/g/signalk/c/uKwSHLbEES4/m/yGD5PPsXCAAJ

FredericGuilbault, 2020-12-12:
As on lysmarine every data flow should go thru signalk, this is the plugin that need to be configured to enable udp nmea communication. 

https://www.npmjs.com/package/@signalk/udp-nmea-plugin

Now the task would be to add it and configure it in the build script.

mgrouch, 2020-12-22:
Close for now

closed 2020-12-22 by mgrouch