lysmarine/lysmarine_gen #34: Package stellarium app and make view point read from IMU to create augmented reality night sky view

mgrouch, 2020-10-13:
For ideas how to script it more info here

https://bareboat-necessities.github.io/my-bareboat/#_celestial_navigation

FredericGuilbault, 2020-11-24:
Im all in for stellarium. But idk for the IMU. 

Not sure how much of a usecase it is to have rpi with a screen not fixed to anything that you can aim at the sky. For the size of the implementing job.

The other main reason why people seem to use IMU on board is autopilot and windvane compensation. Which in both case require the IMU to be bolted and calibrated in place. 

I stay open to receive testamonials and examples of people that have build portable setups tho.

mgrouch, 2020-11-24:
That's the point of IMU. It knows orientation. In fact it needs movement to
self calibrate.
There are many free applications for smartphones you can try. Example "Star
walk".
The idea is to try to reproduce something like that using Stellarium.

FredericGuilbault, 2020-11-24:
For a cellphone it make sense, but for a RPI ? Most of rpi based boat computers I saw on the internet are screwed in place (including yours). I get the idea but IDK how someone would use it in real life. 

Unless we duck-tape the IMU on a Jacob's staff or something :p

FredericGuilbault, 2020-11-24:
Oh, stellarium have a web version,   https://github.com/Stellarium/stellarium-web-engine    maybe this one could use a phone IMU and the phone screen.

mgrouch, 2020-11-24:
It should work (not talking about web version) if your RPI is stationary
and your screen is stationary in the cockpit (let's say in sailpod).

We could read heading from pyPilot (which is using IMU) or from signalK
which gets it from pypilot.

Setup stellarium to connect to GPSD daemon  and control view in Stellarium
using true heading and pitch.
(same way telescope would control it)

mgrouch, 2020-11-25:
Stellarium is packaged in my fork
needs some more scripting 
for augmented reality

FredericGuilbault, 2021-03-23:
At second thought, i don't think ill add stelarium. 

it's kinda outside the scope of the navigational tools ( compared to sextant calculator or ephemeride tables, or a navigational star finder)    ... and it have a so much nicer user experience on a cellphone if it's for learning or fun.

closed 2021-03-23 by FredericGuilbault