lysmarine/lysmarine_gen #37: Better default setting and styling of OnBoard on screen keyboard

mgrouch, 2020-10-16:
Better default setting and styling of OnBoard on screen keyboard
to match the default desktop theme, fonts, etc

mgrouch, 2020-11-10:
Use same Sans font for key labels as on desktop

FredericGuilbault, 2020-11-17:
There is a missing newline in ./config/openbox/autostart that break the config injection for onboard. 

This is what you should see: 
[image]

are you seeing this or the fullscreen whit one ?

mgrouch, 2020-11-25:
Done in my fork

FredericGuilbault, 2020-12-02:
Related commits found by digging git-blame :

https://github.com/bareboat-necessities/lysmarine_gen/commit/eb6d5ab93d62279ca8222218bb356f837999c6a1
https://github.com/bareboat-necessities/lysmarine_gen/commit/b97fa81be79ec96da893fd1f5a7bd80925a55af2
https://github.com/bareboat-necessities/lysmarine_gen/commit/0d61409ee2de3d961f36d6bc96da08efd7f98be7
https://github.com/bareboat-necessities/lysmarine_gen/commit/0d61409ee2de3d961f36d6bc96da08efd7f98be7

FredericGuilbault, 2020-12-14:
fixed here : 
fc6606cadda0b603f93ef7a755992342840ca493
0b3b9ea4e962814a6738b5f94217bcc491e281e4

closed 2020-12-14 by FredericGuilbault