lysmarine/lysmarine_gen #38: OpenCPN needs better touch gestures support

mgrouch, 2020-10-16:
Patch exists.

See
https://github.com/OpenCPN/OpenCPN/issues/2057

Hopefully it will make it into OpenCPN when WxWidgets gets to 3.2 release.

Thanks

mgrouch, 2020-10-18:
I've created binaries for evedev-rce on launchpad.

https://launchpad.net/~bbn-projects/+archive/ubuntu/bbn-ppa

mgrouch, 2020-10-23:
Ubuntu packages for 

https://launchpad.net/~bbn-projects/+archive/ubuntu/bbn-ppa/+packages

amd64
arm64
armhf
ppc64el

Groovy, Focal, Bionic

mgrouch, 2020-11-08:
I've built few lysmarine images with my patched version of OpenCPN with wxwidgets 3.1
and my patch for touch.
I had to retract from it. my build of OpenCPN (and wxWidgets) was using GTK3 which already had client side window
decorations only. Since we are running it in fbpanel GTK2 window manager we lost all window decorations.

mgrouch, 2020-11-25:
Done in my fork

FredericGuilbault, 2020-12-12:
I asked on openCPN and team member said it will be integrated in Ocpn. As this is a feature and not an urgent bug fix. and as it's just a question of delay. ill Prefer bundling the original project.

closed 2020-12-12 by FredericGuilbault