lysmarine/lysmarine_gen #45: Set up automatic periodic builds on one of public CI services

mgrouch, 2020-10-16:
Set up automatic  periodic builds on one of public CI services

mgrouch, 2020-10-19:
See example

https://disconnected.systems/blog/custom-rpi-image-with-github-travis/

mgrouch, 2020-10-22:
Or with CircleCI and cloudsmith. 
Example

https://github.com/bareboat-necessities/evdev-right-click-emulation/tree/hold-right-click/.circleci

mgrouch, 2020-10-24:
I've started working on that. It runs on circularci
However it fails in 
libs.sh 
part 

		log "Resize the filesystem to fit the partition."
kpartx 

where it tries to find loop ID.

circleci scripts are here

https://github.com/bareboat-necessities/lysmarine_gen/tree/master/.circleci

FredericGuilbault, 2020-10-29:
I have compared your branch with mine this morning. You made a lot of work and fast !  As I understand it, everything CircleCI related is located in `./.circleci/` and there are other changes that had to be made at various location in order to make lysmarine_gen build on CircleCI.
 
But I also found changes that are out of the scope of CircleCI integration (and one that I disagree with). So that's kinda won't be possible to simply make a pull request straight from your master. So, IDK what would be the best way to integrate what you have done into lysmarine's repo.

For my self. From now on, I will work on branches instead of master. This will avoid a lot of merge conflicts.

mgrouch, 2020-10-29:
The changes you need are in
.circleci
libs.sh
raspbian.sh

(Mounting, unmounting, temp files cleanups)

There are also some temp files clean ups in stages scripts which were leaving files after and arm64 build would run out of space. 

it’s pretty straightforward ones you start integrating changes and run them in circle CI.

mgrouch, 2020-10-29:
Btw I’ve tried to load both images 32/64. They both load fine on my pi4.

FredericGuilbault, 2020-10-29:
This is the branch : 
https://github.com/lysmarine/lysmarine_gen/tree/CircleCI

FredericGuilbault, 2020-10-29:
@mgrouch 
in ` config.yml` there is a block that change `install-scripts` location:

```
commands:
  git-checkout:
    steps:
      - checkout
      - run: mv install-scripts cross-build-release/
```
And accordingly, `raspbian.sh` script have been adapted. But it break local build.  

Idk the reason for this, can we move it the repo... Or not at all.
A 3rd way could be to move it with `raspbian.sh` ?  So both CircleCI and local build will behave the same.

mgrouch, 2020-10-29:
You can expand replace checkout command
- run: git ...
And have full control where the code is checked out to.

mgrouch, 2020-11-06:
See also my post here

https://www.cruisersforum.com/forums/f134/circleci-builds-for-arm-opencpn-242123.html

FredericGuilbault, 2020-11-10:
I was trying to make a little cleanup and warp my head around config.yml before integrating armbian, debian and vbox debian.  So I could close this issue :D 

 Then I realized that circleCI does not support ARM arch at all ?!? 

You sure made an impressive hack of docker here :+1:  @mgrouch . Just to know before I grind my teeth on this. Have you tried to use buildx first ?

mgrouch, 2020-11-10:
> You sure made an impressive hack of docker here 👍 @mgrouch

Thanks! I haven't tried buildx. But I already made arm32/64, amd64 builds
for lysmarine, and a number of its components on CircleCI (with qemu
docker) deployable to CloudSmith.
Take a look at my forks under bareboat-necessities github. Kplex,
evdev-rce, lysmarine_gen, fbpanel, gdkpixbuf, gtk3, wxWidgets3.1, OpenCPN,
wxsvg3.
Unfortunately I realized GTK3 version of opencpn is too bleeding edge for
GTK2 desktop to handle. So I need to set up GTK2 and wxWidgets3.1 with
OpenCPN for GTK2 build.
I've also added mopidy music player to lysmarine image. And evdev-rce you
definitely need to have for a touch screen.

FredericGuilbault, 2020-11-12:
I made changes on the build configs of circleCI AND I also busted my free plan. 

In all case, the build is now considerably slower than it was at first. 

[image]

I suppose you also have buster your plan too, what is your build time ?

mgrouch, 2020-11-12:
I use just free plan. 1:40 minutes to build lysmarine for arm64 for me.

mgrouch, 2020-11-12:
And mostly it spends time compiling serial and mdns modules while installing SignalK

FredericGuilbault, 2020-11-15:
It's really because I did not use your container. My problem is that I really can't let go my capability to make local builds. 
When I will have more time I will try to implement it in a way that work both on a local machine and with circleCi. 
I also have a problem with virtualbox that refuse to work on circleCi because they do not support vt-x technologies.

FredericGuilbault, 2020-11-23:
The work is happening in this branch: 
https://github.com/lysmarine/lysmarine_gen/tree/CircleCI

FredericGuilbault, 2020-11-24:
Im trying to reimplement the arm based docker container and im having difficulties to use kpartx from inside the container. 
```
[19:46:04]  Resize the filesystem to fit the partition. 

+ echo ./cache/raspios-arm64/raspios-arm64.base.img-inflated
./cache/raspios-arm64/raspios-arm64.base.img-inflated

+ fdisk -l ./cache/raspios-arm64/raspios-arm64.base.img-inflated
Disk ./cache/raspios-arm64/raspios-arm64.base.img-inflated: 8 GiB, 8589934592 bytes, 16777216 sectors
Device                                                 Boot  Start      End  Sectors  Size Id Type
./cache/raspios-arm64/raspios-arm64.base.img-inflated1        8192   532479   524288  256M  c W95 
./cache/raspios-arm64/raspios-arm64.base.img-inflated2      532480 16777215 16244736  7.8G 83 Linu

+ kpartx -l ./cache/raspios-arm64/raspios-arm64.base.img-inflated
mount: could not find any device /dev/loop#Bad address

can't set up loop

Exited with code exit status 1
CircleCI received exit code 1
```

mgrouch, 2020-11-24:
It didn’t work for me either. That’s why I had to switch to mounting it without kpartx. I think my code is more generic. You can just modify my code to make it run locally. You will be able to build arm images even on your PC. BTW what I have in my fork right now is quite a milestone version. I even tagged it. And posted image.

mgrouch, 2020-11-24:
There are so many improvements in my fork. You should try getting it as is, replace cloudsmith location to yours in my config.xml and build it too. It builds fine rpi4 arm64. Please take a look, load the image. Lots of Issues I opened are fixed in it.

FredericGuilbault, 2020-11-24:
> That’s why I had to switch to mounting it without kpartx.

https://github.com/bareboat-necessities/lysmarine_gen/blob/046e5e07496dd947d32bf30a972abacfb71289c8/cross-build-release/lib.sh#L102

That's not what your code says. 
---
> You can just modify my code to make it run locally 

Yeah that's what im trying to do ATM, I tried to copy-paste what you did, but when I saw that is was breaking the localbuild I had to roll back, then I make everything work without docker, and now in trying to reimplement your docker container to gain build speed. But no luck so far :(

---
> You should try getting it as is. 

You choose to fork the project instead of contributing. Im totally ok with this. Im inspired by all the changes you do and idea you bring im slowly implementing some. I find it cool that you can go fast and try all the features you want go back and forth without having to justify and debate decisions. But it's your fork now and in not looking for an other FOSS project to contribute.

mgrouch, 2020-11-24:
I could only do it in my fork cause I needed CircleCi and Cloudsmith. I wouldn’t not have access to them. What started and idea of making small contributions turned a bit into deviation I was carried away with experimental changes. But they do look good and I’d wish them to me it in mainline.

mgrouch, 2020-11-24:
Regarding kpartx. I guess I forgot how I got it working. I remember 
adding -s overlay2
was really important. 
the combination of versions of VM and dockers
was important. And some other options 
I pass to docker commands.
I really do not know how to make it work without
dockers though.

FredericGuilbault, 2020-11-24:
Solved by ea69914d3bb63f712e709275e6f4d5eb99e42952 
For an unknown reason mounting /dev in the container solved it :)

mgrouch, 2020-11-24:
I’ve created a pull request for everything I’ve done. That’s the first version I’ve built that I was happy about.

mgrouch, 2020-11-24:
Btw why are you building for so many distros? It’s hard to maintain give the resources. 

I think arm64 and amd64. 
Arm64 raspi OS based
And amd64 do not know.  Debian?

you need to multiple these 2 
by another 2. Buster and bullseye

and it’s already 4 a lot to handle.

FredericGuilbault, 2020-11-25:
yeah.... :( 

I like armbian, it supports ALL the SBC from a single OS. it seems like the easy choice ....except that they do not support RPI.

So I need to also bundle raspbian.... but now come in 2 arch. 

Debian is a must, to reach navigatrix users and the "I have this old laptop" usecases, .... But I had trouble building it was messy. In the last week I finally managed to build a liveOS with the installer. so im currently cleaning all this mess.   

Finally, Debian in virtualbox is the fastest dev cycle I used so far  hack->build->test->repeat under 30 min

So we have 
 - armbian(based on DebianBuster + compatibility layers ) (that build only pineA64so so far.)
 - raspios 32 (based on DebianBuster + pi toolset )
 - raspios 64 (based on DebianBuster + pi toolset )
 - DebianLive ( DebianBuster) (with installer)

and the following are going away
 - raspbian 
 - debian livebuild 
 - debian.IMG 
 - and debian-VBOX as a release version. But will remain as a dev tool.  
 
I won't be supporting both Buster and bullseye there is no user reach gain in doing this.

mgrouch, 2020-11-25:
Looks good. Will the first feature complete version be available by Christmas?

FredericGuilbault, 2020-11-25:
I think so, im quite close, I have merged a big chunk this morning.  Some debugging left but the bases are there.

FredericGuilbault, 2020-11-27:
I constantly lack of space on Sourceforge, And it marks the build as failed. 

They only provide 10G of storage  this is quite low to store 4 Operating systems (armbian, raspios32. raspios64, debian-live) 
and their first tier is 30G for 100$/month. 

That might not be a livable option after all..... I wish I would have check that sooner.

mgrouch, 2020-11-27:
I think the last step of CircleCi could upload it somewhere else.

mgrouch, 2020-11-27:
I’ve seen people suggest 

https://archive.org/details/open_source_software?&sort=-downloads&page=2

But I haven’t tried

FredericGuilbault, 2020-12-14:
This is done and working. Thx a lot for the help on this one @mgrouch 

https://lysmarineos.com/#/doc/download

closed 2020-12-14 by FredericGuilbault