lysmarine/lysmarine_gen #49: Add script for downloading free charts if avail based on user city selection for timezone

mgrouch, 2020-10-23:
Add script for downloading free charts if avail based on user city selection for timezone.

So that after installation OpenCPN would boot already with the charts.

Thanks

FredericGuilbault, 2020-10-27:
That would be nice ! The best would be to make it as a signalk or opencpn plugin. So anyone could profit from it. it's always more collaborative when problems can be solved upstream ( and less maintenance work afterwards for me ;)  ).

mgrouch, 2020-11-08:
Well. With long touch right click emu downloading charts for a given region is actually
simple already. In OpenCPN chart downloader plugin you can just hold touch in a grid for selected region and gives option
to 'select all'

closed 2020-11-25 by FredericGuilbault

mgrouch, 2020-12-02:
related https://github.com/OpenCPN/OpenCPN/issues/2114