lysmarine/lysmarine_gen #53: Use prebuilt kplex for 64 bit version.

mgrouch, 2020-11-03:
I've put kplex to build via circleCI and upload to cloudsmith

Binaries are here:

https://cloudsmith.io/~bbn-projects/repos/bbn-kplex/packages/

FredericGuilbault, 2020-11-21:
Im working on this in the kplex repository  and their google-group, They need better debianization. At least something good enough for stripydog to host it publicly and good enough to be able to compile on launchpad. 

https://github.com/stripydog/kplex/issues/48
https://github.com/stripydog/kplex/pull/47

mgrouch, 2020-11-25:
Done in my fork

FredericGuilbault, 2020-12-14:
https://github.com/stripydog/kplex/issues/50
https://github.com/stripydog/kplex/pull/46#issuecomment-742950353

FredericGuilbault, 2021-01-16:
I will use a version from my repo & ppa, untill stripydog merge the pull request. Then I will switch to original repository, and maybe even official PPA.

closed 2021-01-16 by FredericGuilbault