lysmarine/lysmarine_gen #57: mdns and serialport used by signalK do not have precompiled binaries for arm

mgrouch, 2020-11-08:
mdns and serialport used by signalK do not have precompiled binaries for arm

as a result they are getting compiled everytime signalK is installed.

FredericGuilbault, 2020-11-21:
For runtime, it make no differance, but for development.  that's quite problematic.    On the other hand IDK what to do in the scope of lysmarine for this. Maybe some caching. but the real solution would be to help/convince mdns people to prebuild for arm... 

Im sure people working on signalK suffer from the same problem. IK they were when sqlite3 had the same problem. I wonder how they deal with this.

mgrouch, 2020-11-22:
Anyone can upload binaries into npm repository. I think it can be easily done using my approach on CircleCi with qemu and arm raspiOS dockers. 
I just haven’t gotten to it yet. Never built npm packages either so far.

FredericGuilbault, 2020-12-14:
In all case this won't be fixed as a part of lysmarine_gen.

closed 2020-12-14 by FredericGuilbault