lysmarine/lysmarine_gen #58: Recognize more serial port USB devices. For example dAISy AIS

mgrouch, 2020-11-08:
Recognize more serial port USB devices. For example dAISy AIS.
I see some configs to recognize some known GPS chipsets.
How do we recognize dAISy AIS serial ports and other NMEA ports via FTDI (those could be anything wind/depth)?

mgrouch, 2020-11-08:
I do like approach based on vendor/product ID. It lets you to re-plug device into different USB port and still be named same

FredericGuilbault, 2020-11-25:
gpsd provide a nice Udev list of known GPS device, and with udev event. I been able to dynamically add GPS signal to signalk and even reboot SK server on new device. That's the Approach I like the most. 

Sadly such lists are not very easy to find. Plus on many cases when it's DIY stuff you only see that data are coming for let's say I2C, 1Wire or TTL      only the human will be able to say if it's dept sensor, a temp sensor or a wind vane that are plugged on the other side. 

That's why people still use node-red and openplotter....

mgrouch, 2020-11-25:
You can do it by vendor And product ids

mgrouch, 2020-11-25:
You can see for AIS here

https://bareboat-necessities.github.io/my-bareboat/images/config/openplotter-serial-devices.png

mgrouch, 2020-11-25:
https://www.cruisersforum.com/forums/f13/daisy-open-source-ais-receiver-154962-3.html

mgrouch, 2020-12-15:
I've done some changes in my fork. I know only vendorID/productID for dAISy AIS. 
And I can't test my changes cause it's on my winterized boat.

FredericGuilbault, 2021-03-30:
https://github.com/bareboat-necessities/lysmarine_gen/blob/master/install-scripts/4-server/files/90-lys-ais.rules
https://github.com/bareboat-necessities/lysmarine_gen/blob/master/install-scripts/4-server/files/manage_ais.sh
https://github.com/bareboat-necessities/lysmarine_gen/blob/master/install-scripts/4-server/files/settings.json