lysmarine/lysmarine_gen #64: Armbian fail to build

FredericGuilbault, 2020-11-12:
Armbian fail to build  

Reported here: 

https://forum.armbian.com/topic/15028-bugi-think-linux-dtb-current-sunxi64_2008_arm64deb-error-while-loading-varlibdpkgtmpcipreinst-exec-format-error/

mgrouch, 2020-11-12:
These allwinner board based car dashboards look interesting 
Could be used for a boat?

https://m.alibaba.com/product/62469626640/4G-64G-IPS-Screen-Car-DVD.html?__sceneInfo=%7B%22type%22:%22appDetailShare%22,%22data%22:%7B%22filterKey%22:%227.8.2_share%22,%22cacheTime%22:%221800000%22%7D%7D

FredericGuilbault, 2020-11-15:
I  Reported it on GitHub too since there was no response on the forum.   https://github.com/armbian/build/issues/2349

FredericGuilbault, 2020-12-01:
solved by a conjuncture of this commit 46a00b2a8fb6932b46911c7b5ced1bb9250f458d and upstream that got his availability back

closed 2020-12-01 by FredericGuilbault