lysmarine/lysmarine_gen #68: speedSample build is now failing

mgrouch, 2020-11-17:
npm WARN deprecated signalk-client@0.0.1-4: Please use @signalk/client instead.
[            ......] / install:proxy-addr: info lifecycle proxy-addr@2.0.6~inst
> mdns@2.7.1 install /var/www/speedSample/node_modules/mdns
> node-gyp rebuild

make: Entering directory '/var/www/speedSample/node_modules/mdns/build'
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_sd.o
../src/dns_sd.cpp:140:49: error: operator '||' has no right operand
 #if defined(kDNSServiceErr_BadSig) || _DNS_SD_H >= 1610100
                                                 ^~
../src/dns_sd.cpp:143:49: error: operator '||' has no right operand
 #if defined(kDNSServiceErr_BadKey) || _DNS_SD_H >= 1610100
                                                 ^~
../src/dns_sd.cpp:146:52: error: operator '||' has no right operand
 #if defined(kDNSServiceErr_Transient) || _DNS_SD_H >= 1610100
                                                    ^~
../src/dns_sd.cpp:149:60: error: operator '||' has no right operand
 #if defined(kDNSServiceErr_ServiceNotRunning) || _DNS_SD_H >= 1610100
                                                            ^~
../src/dns_sd.cpp:152:68: error: operator '||' has no right operand
 #if defined(kDNSServiceErr_NATPortMappingUnsupported) || _DNS_SD_H >= 1610100
                                                                    ^~
../src/dns_sd.cpp:155:65: error: operator '||' has no right operand
 #if defined(kDNSServiceErr_NATPortMappingDisabled) || _DNS_SD_H >= 1610100
                                                                 ^~
../src/dns_sd.cpp:158:51: error: operator '||' has no right operand
 #if defined(kDNSServiceErr_NoRouter) || _DNS_SD_H >= 1710400
                                                   ^~
../src/dns_sd.cpp:161:54: error: operator '||' has no right operand
 #if defined(kDNSServiceErr_PollingMode) || _DNS_SD_H >= 1710400
                                                      ^~
../src/dns_sd.cpp:164:50: error: operator '||' has no right operand
 #if defined(kDNSServiceErr_Timeout) || _DNS_SD_H >= 3200500
                                                  ^~
In file included from ../src/mdns.hpp:11,
                 from ../src/dns_sd.cpp:1:
/root/.cache/node-gyp/10.23.0/include/node/node.h:573:43: warning: cast between incompatible function types from 'void (*)(v8::Local<v8::Object>)' to 'node::addon_register_func' {aka 'void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)'} [-Wcast-function-type]
       (node::addon_register_func) (regfunc),                          \
                                           ^
/root/.cache/node-gyp/10.23.0/include/node/node.h:607:3: note: in expansion of macro 'NODE_MODULE_X'
   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
   ^~~~~~~~~~~~~
../src/dns_sd.cpp:322:1: note: in expansion of macro 'NODE_MODULE'
 NODE_MODULE(dns_sd_bindings,node_mdns::init);
 ^~~~~~~~~~~
In file included from /root/.cache/node-gyp/10.23.0/include/node/node.h:63,
                 from ../src/mdns.hpp:11,
                 from ../src/dns_sd.cpp:1:
/root/.cache/node-gyp/10.23.0/include/node/v8.h: In instantiation of 'void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = node::ObjectWrap; T = v8::Object; typename v8::WeakCallbackInfo<P>::Callback = void (*)(const v8::WeakCallbackInfo<node::ObjectWrap>&)]':
/root/.cache/node-gyp/10.23.0/include/node/node_object_wrap.h:84:78:   required from here
/root/.cache/node-gyp/10.23.0/include/node/v8.h:9502:16: warning: cast between incompatible function types from 'v8::WeakCallbackInfo<node::ObjectWrap>::Callback' {aka 'void (*)(const v8::WeakCallbackInfo<node::ObjectWrap>&)'} to 'Callback' {aka 'void (*)(const v8::WeakCallbackInfo<void>&)'} [-Wcast-function-type]
                reinterpret_cast<Callback>(callback), type);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/.cache/node-gyp/10.23.0/include/node/v8.h: In instantiation of 'void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = Nan::ObjectWrap; T = v8::Object; typename v8::WeakCallbackInfo<P>::Callback = void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)]':
../../nan/nan_object_wrap.h:65:61:   required from here
/root/.cache/node-gyp/10.23.0/include/node/v8.h:9502:16: warning: cast between incompatible function types from 'v8::WeakCallbackInfo<Nan::ObjectWrap>::Callback' {aka 'void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)'} to 'Callback' {aka 'void (*)(const v8::WeakCallbackInfo<void>&)'} [-Wcast-function-type]
make: *** [dns_sd_bindings.target.mk:165: Release/obj.target/dns_sd_bindings/src/dns_sd.o] Error 1
make: Leaving directory '/var/www/speedSample/node_modules/mdns/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:191:23)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.15.0-1027-gcp
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /var/www/speedSample/node_modules/mdns
gyp ERR! node -v v10.23.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN rollback Rolling back bindings@1.2.1 failed (this is probably harmless): /var/www/speedSample/node_modules/bindings is not a child of /usr/lib
npm WARN rollback Rolling back nan@2.14.2 failed (this is probably harmless): /var/www/speedSample/node_modules/nan is not a child of /usr/lib
npm WARN rollback Rolling back mdns@2.7.1 failed (this is probably harmless): /var/www/speedSample/node_modules/mdns is not a child of /usr/lib
npm WARN bootstrap@4.5.3 requires a peer of popper.js@^1.16.1 but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! mdns@2.7.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mdns@2.7.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /tmp/empty-cache46/_logs/2020-11-17T09_38_19_539Z-debug.log

mgrouch, 2020-11-17:
And BTW why is speedSample still on gitlab instead of github as others?

https://gitlab.com/FredericGuilbault/speedSample

Thanks

FredericGuilbault, 2020-11-25:
No real reason, that artifacts of the past. I made speedSample as a class exercise while I was learning nodejs(it was a mistake, I should have learned python)    before even working lysmarine. 

I must I personally use it for fine trimming of sails. But it should be a signalK app available for every OS. And last year I started to use SailFreeGPS app on the cellphone... so IDK if it's still reverent to package it.

mgrouch, 2020-12-15:
I've dropped it from my build completely

closed 2020-12-15 by FredericGuilbault