lysmarine/lysmarine_gen #72: MenuButtons and touch do not work sporadically

mgrouch, 2020-11-18:
Seems to be GTK2 issue. I see it in many apps (all GTK2, ex: fbpanel, opencpn, xfce4-panel)
with hamburger icons.
tint2 is not impacted with this issue (tint2 is not GTK based)

mgrouch, 2020-11-18:
Reported here

https://gitlab.gnome.org/GNOME/gtk/-/issues/3372

mgrouch, 2020-11-20:
Fbpanel has to go. As long as I do not touch anything on fbpanel things work fine with touch screen. 
After few touches on fbpanel things break. Menus do not show up. Can’t switch tabs. 
I think we should have lxpanel instead. I’ve played with tint2, xfce4-panel and they all had issues too. 
lxpanel is what raspberry OS uses and I do not have any issues with touch events on it, even with arm64

mgrouch, 2020-11-20:
And btw lxpanel is successor of fbpanel

mgrouch, 2020-11-22:
After more test it all boils down to gtk2. We need to use OpenCPN compiled with gtk3

mgrouch, 2020-11-25:
Opened an issue in OpenCPN

FredericGuilbault, 2020-12-14:
since I moved to lxpanel-gtk3 it didn't happen to me. is it happening under budgie with gtk2 apps ?

mgrouch, 2020-12-15:
gtk2 apps are the ones which seems to break it. even for other apps (gtk3). qt apps always work though with touch fine

FredericGuilbault, 2021-03-30:
Solved by moving to awesomeWM

closed 2021-03-30 by FredericGuilbault