lysmarine/lysmarine_gen #73: In onboard setting "Shrink work area" needs to be defaulted to false

mgrouch, 2020-11-18:
In onboard setting "Shrink work area" needs to be defaulted to false

FredericGuilbault, 2020-11-18:
Im curious why, What is the impact ? Or the problem it solves

mgrouch, 2020-11-19:
Try opening OpenCPN settings dialog. It’s going to end into endless loop of shrinking and expanding.

mgrouch, 2020-11-19:
Keyboard is movable with the finger anyway. There is that button with arrows pointing 4-way. You can drag keyboard by that icon

mgrouch, 2020-11-25:
Done in my fork

FredericGuilbault, 2020-11-29:
Hummm I can't reproduce it, Is it happening after you made some change to the window decorations or this changes on your fork ?
 
https://github.com/lysmarine/lysmarine_gen/issues/61

mgrouch, 2020-11-30:
It was very reproducible, by opening OpenCPN settings dialog with touch with no keyboard connected

mgrouch, 2020-12-22:
Done in my fork

closed 2020-12-22 by mgrouch