lysmarine/lysmarine_gen #74: WiringPi arm64 deb package needed

mgrouch, 2020-11-19:
WiringPi arm64 deb package needed

FredericGuilbault, 2020-12-14:
ewwww, this package appears to be a complete clusterfuck of bitterness and despair.  

Official statement:     http://wiringpi.com/wiringpi-deprecated/
And the actual mess :   https://github.com/RPi-Distro/repo/issues/214   

And this comment is calling for a technical deprecation from kernel 5.5
https://github.com/WiringPi/WiringPi/issues/80#issuecomment-731371793

From here it looks( I haven't tested ) that wiringpi arm64 don't simply need compiling but a dedicated repository.  
https://github.com/Vicinzu/wiringPi-arm64
https://github.com/WiringPi/WiringPi/issues/82#issuecomment-739616069

___
Also, the last package that guy distributed have effectively no source code available so, yes. It's factually not FOSS and AFAIK this prevents everyone tied to the GPL to package it legally.

mgrouch, 2020-12-22:
Let’s close it then

closed 2020-12-22 by mgrouch