lysmarine/lysmarine_gen #81: App menu needs Reorg after migrating to budgie

mgrouch, 2020-11-25:
App menu needs Reorg after migrating to budgie.
Just config issue

mgrouch, 2020-11-25:
Just want to reiterate that budgie desktop is awesome and it’s way to go

FredericGuilbault, 2020-11-25:
lolllll got it bro XD

mgrouch, 2020-12-01:
It needs to be set up explicitly. As soon as you inherit base ones you end up with a mess in menus.
It does mean that for every application we need .desktop file in user home directory (.local/share/applications).
I suggest having category prefixed "LysCat_" for them too so it's easier to recognize.

FredericGuilbault, 2020-12-01:
> It does mean that for every application we need .desktop file in user home directory (.local/share/applications).
if the application are not installed under in the /home/$USER/.local/ directory neither .desktop file should. 
/usr/local/share/applications would probably be a better place.

mgrouch, 2020-12-01:
Will it get overwritten on 
apt update; apt upgrade?

FredericGuilbault, 2020-12-01:
not if it's a /local/ directory, 

for example: apt manage everything in /usr/bin and manual install should be in /usr/local/bin so apt can't override it.

mgrouch, 2020-12-02:
I think I got it solved in my fork

closed 2020-12-02 by FredericGuilbault