lysmarine/lysmarine_gen #87: All the chartplotters need to be setup to use same chart directory to re-use charts

mgrouch, 2020-11-25:
All the chartplotters need to be setup to use same chart directory to re-use charts

FredericGuilbault, 2020-11-26:
yes, in the folder `/srv/charts` 

The folder /home `/user/charts` bind to `/srv/charts` OpenCpn is pre-configured to use this folder, 
freeboard-sk rely on the sk charts server which is instructed to store/query mbtitles charts from `/srv/charts`

There is a known problem with having xygrib and opencpn both packaging the same costal line seperately, I contacted both regarding the issue and the answer where :

xygrib: we lack of resource to make it append. 
opencpn: we don't see that as a problem,  windows users don't use package managers. Their concern was mainly about having the charts in their git repo or not. 

is there some other place where charts are needed ?

mgrouch, 2020-12-15:
They need to have single place for original downloaded charts which we have I think. OpenCpn one.
All signalK based ones will use it as primary source to generate mbtiles from it.
Then there is AvNav and qtVlm. they also should use it as primary local source.
AvNav also generates mbtiles from it.
qtVlm I need to take a closer look at what it does

FredericGuilbault, 2020-12-15:
> They need to have single place for original downloaded charts which we have I think. OpenCpn one.

Each app have their location... that's kinda the core of the problem. 

>All signalK based ones will use it as primary source to generate mbtiles from it.

no, the signalk user should not have read/write access to the `user` home.

When AvNav will be implemented, as it's a web server with write capability. It will have its own user and should have the most limited access to other users privileges. 

No app should be the god app that hold the canonical location where all others app should bind to. specially if this live app only in the userspace. 

In lysmarine. The core point of the decision of using `/srv/charts` was. 
- A location that is app neutral (so app depend on another app location/existence)
- A location that is user neutral (cuz many users will need to access them: User, signalk, www-data, root )
- A location compliant to the FHS 
- A location that is historically used to store large datasets of dynamic  content. 
- A location where human users could manage the dataset without root access. 

Decision made considering that most apps have a config file where we can change the location, or we can link folders in the worst cases. It's possible to create a single location without having to sit every upstream at a table until they decide for a standard (cuz this likely won't happen)

___
For the record `/var/share/charts` was my first guess. But after discussion on IRC with hardcore linux devs  `/srv/charts` was found more appropriate.