lysmarine/lysmarine_gen #90: For longer plan desktop launcher needs improvements

mgrouch, 2020-11-25:
Ability to launch native apps
Not be web based. Web based takes lots of memory
(I've seen some written in Vala. https://aur.archlinux.org/packages/slingswarm/  Needs to forked and re-worked)
And integrate it with NMEA data feed, AIS
Make it look ala https://bareboat-necessities.github.io/my-bareboat/cockpit-ui/cockpit-ui.html

FredericGuilbault, 2020-11-26:
nativfier and nodejs eat the memory cuz they are both based on the chromeV8 engine and it don't share resources, 

This "electronJs all the things" trend of the past years is a total cancer.... but it's 2020 what Can I do... 

Web based stuff don't waste resources, running 10 V8 engines on the same RPI do.

FredericGuilbault, 2020-12-14:
Have try it, could it replace the panel bar ?

mgrouch, 2020-12-14:
It’s fully implemented in my fork. 
Look for bbn-launcher

FredericGuilbault, 2020-12-14:
I saw your last video and you still had the bar at the top. So I guess it is not replacing it.

mgrouch, 2020-12-15:
It's replacing it for touch, I just like the final distro be a blend of desktop and touch screen OS.
Android sucks as desktop. Can be done better

mgrouch, 2020-12-22:
Almost finished in my fork

closed 2020-12-22 by mgrouch