lysmarine/lysmarine_gen #91: Add NavTex and Inmarsat Fleet decoder apps

mgrouch, 2020-11-25:
Add NavTex and Inmarsat Fleet decoder apps

FredericGuilbault, 2020-11-26:
I would def accept pull request about those apps.

mgrouch, 2020-12-15:
Well it boils down to proper antenna and proper SDR.
Cheap SDRs do not support the bands needed to NavTex and (might) work with upconverters which drives the cost of the whole solution up anyway. Then those bands antennas to be practical and small enough on a boat need to be active antennas, properly
grounded. The one I've seen was The PA0RDT Mini Whip http://dl1dbc.net/SAQ/miniwhip.html

mgrouch, 2020-12-15:
SDR is very fast developing area. It looks like all human (and possibly alien :) ) RF traffic can be analyzed by software 
and rpi4 is enough for computing power. Antennas and details of the set up reducing interference are important, SDR band range too (but that is something which drops in price rapidly)
Software decoders for all radio transmissions exist already in one shape of from (almost all are free).

FredericGuilbault, 2020-12-15:
Im already sold to the idea, all I need is a a pull request..... cuz on priority lis,t there is no chance I reach this issue before 2027

mgrouch, 2020-12-22:
NavTex decoding app added in my fork