lysmarine/lysmarine_gen #100: Xorg server crash with seg fault on some OsColor related funtion call

mgrouch, 2020-12-01:
Xorg server crash with seg fault on some OsColor related funtion call.
Unfortunately I've rebooted pi and lost whole error message

mgrouch, 2020-12-01:
I think Xorg server needs to be automatically restarted on non repeatable crashes,
otherwise we end up with text console login screen with no keyboard in a cockpit (only touchscreen)

mgrouch, 2020-12-02:
[    30.013] (II) modeset(0): Disabling kernel dirty updates, not required.
[ 40481.775] (EE)
[ 40481.775] (EE) Backtrace:
[ 40481.782] (EE) 0: /usr/lib/xorg/Xorg (OsLookupColor+0x1a8) [0x5579c46410]
[ 40481.782] (EE)
[ 40481.782] (EE) Segmentation fault at address 0x78
[ 40481.782] (EE)
Fatal server error:
[ 40481.782] (EE) Caught signal 11 (Segmentation fault). Server aborting
[ 40481.783] (EE)
[ 40481.783] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[ 40481.783] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[ 40481.783] (EE)
[ 40481.783] (II) AIGLX: Suspending AIGLX clients for VT switch
[ 40481.843] (EE) Server terminated with error (1). Closing log file.

mgrouch, 2020-12-15:
Well. Xorg server is not supposed to crash no matter what client is doing.
It's really Xorg bug

FredericGuilbault, 2020-12-15:
yeah, what can be done here is an auto restart (but limited to avoid  a complete cycling that would jam the server level services) 

But there is no way I could patch Xorg my self. This will have to be reported to the Xorg.

FredericGuilbault, 2021-01-16:
start x now restart automatically after 3 seconds on failure.   For the rest you will have to check with the Xorg team.

closed 2021-01-16 by FredericGuilbault