lysmarine/lysmarine_gen #95: Too many postinstall steps to get charts displaying in AvNav, Freenboard

mgrouch, 2020-12-01:
I've downloaded raster charts via opencpn chart downloader plugin
they should have been automatically picked up by freenboard-sk and avnav
(and qtvlm if it's installed)

FredericGuilbault, 2020-12-01:
First questions are where does opencpn chart downloader store the charts ?  
is it configurable ? 
and does freenboard-sk even support raster charts ?

For the two other softwares (avnav and qtlvm)    lysmarine does not package or support them ATM....

mgrouch, 2020-12-15:
freenboard-sk supports only raster charts. All other chatplotters except of OpenCPN are mbtiles based
and currently support only raster mbtiles.
OpenCPN is not tiling plotter but it supports lots of chart formats natively

FredericGuilbault, 2020-12-15:
Sorry my question was off set, the point is: all the charts should be stored in `/srv/charts`    `/home/signalk` and `/home/user` should have a link to this location and have the perms to read and write on it.

If something don't go in there. This need to be fixed.

FredericGuilbault, 2021-01-17:
I checked that. There is a typo in the opencpn config. The location to store charts use "/home/user/Charts" where the correct path is "/home/user/charts" 

Then, opencpn store dataset in subfolders  EX:  "/home/user/charts/Raster/NOAA"    But the charts plugin of signalk do not check recursively in the subs folders.   This will have to be fixed in the charts plugin.