lysmarine/lysmarine_gen #97: Pypilot writes excessive log in /var/log/daemons.*

mgrouch, 2020-12-01:
Pypilot writes excessive log in /var/log/daemons.*
Repeatedly reporting same condition

FredericGuilbault, 2020-12-01:
yeah...... the non-stop logs emitting problem should be addressed at https://github.com/pypilot/pypilot/issues 

As for what we do with this, it's probably possible to send them to /dev/null from the .service file instead of writing them on the disk.

FredericGuilbault, 2020-12-14:
https://github.com/pypilot/pypilot/issues/69

FredericGuilbault, 2020-12-14:
We could  ` > /dev/null` all that crap strait from the .service file but that would be bad practice as that would set us blind in case of real issue....

mgrouch, 2020-12-15:
Seems not even pypilot issue but rather underlying library used by it

FredericGuilbault, 2020-12-15:
it's out of my personal scope to dig too much in pypilot code but, One of my guess is that rtimulib2 keep sending logs cuz pypilot keep calling it.    The other point is that it would be pypilot job to filter rtimulib2 /dev/null. Cuz at our level, if we do it. it;s everything pypilot produce that goes to the void.    

but yeah, if pypilot doen not keep calling rtimulib2 on a infinite loop. the issue should be transfered to rtimulib2 project. 

Now the question that concern lysmarine here, is what we do with these logs ?