lysmarine/lysmarine_gen #98: VNC server fails to start. Password not set error in /var/log/daemon

mgrouch, 2020-12-01:
VNC server fails to start. Password not set error in /var/log/daemon

FredericGuilbault, 2020-12-14:
That's strange, password is hard set in the file:
install -v    -o 1000 -g 1000 $FILE_FOLDER/passwd "/home/user/.vnc/passwd"
https://github.com/lysmarine/lysmarine_gen/blame/master/install-scripts/6-desktop/8-remoteDesktopAccess.sh

And I can't reproduce it. On the contrairy it says that it's using the password file 
[image]

FredericGuilbault, 2020-12-14:
@mgrouch  is this something that happens only on your branch ?

mgrouch, 2020-12-15:
It doesn't seem to happen anymore for me. Can be closed

closed 2020-12-15 by mgrouch