lysmarine/lysmarine_gen #106: Fill empty disk space with zeros before compressing the image to reduce the final size

mgrouch, 2020-12-15:
Fill empty disk space with zeros before compressing the image to reduce the final size.
See:

Something like this:

https://unix.stackexchange.com/questions/44234/clear-unused-space-with-zeros-ext3-ext4

FredericGuilbault, 2020-12-15:
This guy have no intention of reducing his partition size to fit what's on it. therefor he is looking for the best way to pack all this empty space. 

This is not what I'm doing. After the build, before compressing pishrink use resize2fs and truncate to reduce the partition size to what's on it. The only free space I can think of would be the tiny amount of fragmentation that cleanup stage would induce but AFAIK it's would be under 100 Mb. To this we might add the 5% reserved block space that ext4 was forcing in the area of the physical spinning HDD, not sure this is still done as it wouldn't be revelent on SDcards with NAND . 

Maybe im wrong and this would make a difference. But it would require bench testing or a proof of concept. cuz at first this does not apply to us.,    pshrink don't do it. armbian-build don't do it and if I recall pi-gen don't do it either.

mgrouch, 2020-12-22:
Ok

closed 2020-12-22 by mgrouch