lysmarine/lysmarine_gen #116: ld.so.preload error workaround

gplanansky, 2021-05-04:
The  lysmarine-0.10.0-raspios-arm64.img.xz , on a pi4b, generates the ERROR: ld.so ... message with every shell command.   

user@lysmarine:/etc $ date 
ERROR: ld.so: object ‘/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so’ from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Tue May  4 08:11:03 EDT 2021

A workaround is to comment out the offending line in that file; or, that being the only one in the
lysmaring-0.10.0 image (after booting, I did not check the original image), just zero out the file.  

I gather this image is multi-archtecture and wonder if that's related to the issue.

FredericGuilbault, 2021-05-06:
Thx for reporting. It's nice that you have found a workaround. ill look for a permanent fix.

FredericGuilbault, 2021-05-29:
The error occurs only if the image is build on CircleCI, I have updated a locally build image on cloudSmith as version 0.10.1