lysmarine/lysmarine_gen #119: Boots into CLI, no desktop

deanfourie1, 2022-01-23:
Upon every boot, I get "Desktop Starting" then it just hands at cli.

A manual startx opens the desktop manually. 

Is there a way to start on boot automatically?

Thanks

FredericGuilbault, 2022-01-25:
This is strange, It should start automatically via the systemd [service unit called startx](
https://github.com/lysmarine/service_manager/blob/master/servicemanager/startx.service). 

Have you try  `systemctl start startx` and `systemctl enable startx`   ?

deanfourie1, 2022-01-27:
It does start, I was just being too impatient. Also as per the documentation I see there is a way to skip this and login straight into the X session. Sorry but this is resolved!

closed 2022-11-26 by FredericGuilbault